openyou/emokit #291: Getting almost every quality value as None

JulioEstebanAsiainNeno, 2019-07-30:
Hey!

I'm trying to make this work, but most of the quality values I get are None. and some are nothing which is weird since when I use the EmotivBCI it shows all of the sensors in green.
I'm on w10 using the python version and an emoti epoc model 1.0.

Tried setting Old Model to True but it didn't change much.

Emokit - v0.0.8 SN: SN20120301000664  Old Model: True
+========================================================+
| Sensor |   Value  | Quality  | Quality L1 | Quality L2 |
+--------+----------+----------+------------+------------+
|   F3   | 4247.4242419995 | 3688.48484812 |    None    |    None    |
|   FC5  | 4377.2424238047 | 3581.33333298 |    None    |    None    |
|   AF3  | 4208.7878783670 | 49.4545454496 |  Nothing   |  Nothing   |
|   F7   | 3830.1515147685 | 259.63636361 |  Nothing   |  Nothing   |
|   T7   | 4108.3333329225 | 193.696969678 |  Nothing   |  Nothing   |
|   P7   | 4629.6666662037 | 3861.57575719 |    None    |    None    |
|   O1   | 4574.5454540880 | 4009.93939354 |    None    |    None    |
|   O2   | 4690.4545449855 | 2513.93939369 |    None    |    None    |
|   P8   | 3959.9696965737 | 1936.96969678 |    None    |    None    |
|   T8   | 4728.5757571029 | 3132.12121181 |    None    |    None    |
|   F8   | 4220.6363632143 | 1611.39393923 |    Good    |    None    |
|   AF4  | 4276.2727268451 | 3540.12121177 |    None    |    None    |
|   FC6  | 4223.2121207898 | 3461.81818147 |    None    |    None    |
|   F4   | 4601.3333328732 | 2909.57575728 |    None    |    None    |
|   X    |   42.0   |   N/A    |    N/A     |    N/A     |
|   Y    |   42.0   |   N/A    |    N/A     |    N/A     |
|   Z    |    ?     |   N/A    |    N/A     |    N/A     |
|  Batt  |    72    |   N/A    |    N/A     |    N/A     |
+--------+----------+----------+------------+------------+
|Packets Received:  43266   |  Packets Processed:  43265 |
|   Sampling Rate:   132    |        Crypto Rate:   132  |
+========================================================+

tushartyagi90, 2019-10-18:
Hi,
Did you get any solution/reason for the above? I am also facing the same problem with Epoc+.

JulioEstebanAsiainNeno, 2019-10-18:
Nope

warrenarea, 2019-10-18:
emokit was never upgraded to work with the newer EPOC+ models.
you should check out the CyKIT repository instead.  

https://github.com/CymatiCorp/CyKit

Snail546, 2021-03-11:
> emokit was never upgraded to work with the newer EPOC+ models.
> you should check out the CyKIT repository instead.
> 
> https://github.com/CymatiCorp/CyKit

Thanks so much for this!